PalouseRobosub/robosub_simulator-2018 #7: Simple Mesh to SDF conversion

bkallaher, 2016-12-07:
The collada files for things like the buoys and the gate should be converted to SDF. This should be relatively simple as the models are just cylinders and spheres.

bkallaher, 2016-12-15:
Buoys converted.

bkallaher, 2016-12-15:
[image]

Old is on the right and new is on the left.

skallaher, 2016-12-15:
Why does the yellow buoy look white?

bkallaher, 2016-12-15:
That is the old one. I don't know why.

skallaher, 2016-12-15:
Ah, I just got them backwards. Is there an easy way to make them less neon?

bkallaher, 2016-12-15:
Right now I'm just using the gazebo default colors for red, green, and yellow. I'll mess around with the diffusion, emissive, specular, and ambient values a bit to remove some of the "life" from the colors.

bkallaher, 2016-12-15:
Or find a shader map that will do the trick.

closed 2016-12-16 by bkallaher