PalouseRobosub/robosub_simulator-2018 #8: Hydrophone Simulation

bkallaher, 2016-12-07:
Eventually the hydrophones should be simulated using the absolute positions of the sub and the pinger.

BigDCross, 2016-12-07:
So the best way to do this is probably to make a new hydrophone node that pulls in the subs position on the topic '/position' and the location of the obstacles in the pool on the topic '/obstacles/positions'. It would then find the pinger in the '/obstacles/positions' array and then go from there. 

The pinger, itself, will have to be added to the pool as an object and then the pingers name will have to be added to the robosub/param/obstacles.yaml file.

ryan-summers, 2016-12-13:
This was recently added to the code base in https://github.com/PalouseRobosub/robosub_simulator/pull/15

closed 2016-12-13 by ryan-summers