PalouseRobosub/robosub_simulator-2018 #11: Thruster Timeout

BigDCross, 2016-12-08:
The real thrusters, if they have not received a command in some amount of time (I'm not sure what it actually is), will timeout and stop spinning. This should be implemented in the thruster plugin.

BigDCross, 2016-12-13:
@irwineffect do you know how long the actual thruster timeout is?

irwineffect, 2016-12-13:
Nah I don't. @ryan-summers, do you know?

ryan-summers, 2016-12-13:
From the Maestro manual:

`Timeout: This parameter specifies the duration before which a Serial timeout error will occur. This error can be used as a safety measure to ensure that your servos and digital outputs go back to their default states whenever the software sending commands to the Maestro stops working. The serial timeout error will occur whenever no valid serial commands (or qualifying native USB commands) are received within the specified timeout period. A timeout period of 0.00 disables the serial timeout error. The resolution of this parameter is 0.01 s and the maximum value available is 655.35 s. The native USB commands that qualify correspond to the following methods in the Usc class: setTarget, setSpeed, setAcceleration, setPwm, disablePWM, and clearErrors. Running the Maestro Control Center will not prevent the serial timeout error from occurring, but setting targets in the Status tab or playing a sequence will.`

irwineffect, 2016-12-15:
How does a value of 1 or 0.5 seconds sound?

ryan-summers, 2016-12-15:
0.5 is good

BigDCross, 2016-12-15:
See #21

closed 2016-12-16 by BigDCross