PalouseRobosub/robosub_simulator-2018 #16: Fisheye Camera Simulation

skallaher, 2016-12-12:
Convert the current camera simulation to create a fisheye image and publish on either the camera/left or camera/right topic as a WFOVImage message. Looking at these links:

http://gazebosim.org/tutorials?tut=wide_angle_camera&branch=wideanglecamera
http://sdformat.org/spec?ver=1.6&elem=sensor

Provided by @BigDCross in #2, it could be possible to also simulate the lens of the cameras on the sub using intrinsics measured through calibration.

BigDCross, 2016-12-15:
Ok so fish eye cameras work fine if you follow the tutorial (though gazebo will give you warnings about the sensor type being wideanglecamera. I'm not sure the significance of these). One thing that we will need to do ourselves is either convert the sensor_msgs/Image that the ros camera plugin publishes to a wfovImage or write our own camera plugin.

skallaher, 2016-12-15:
A WFOVImage contains a sensor_msgs/Image. You should be able to simply say `wfov_image.image = image` or the like.

BigDCross, 2016-12-15:
Yea shouldn't be hard. Are you using any of the other metadata in the wfov image?

skallaher, 2016-12-15:
Nope, not currently.

irwineffect, 2016-12-16:
@skallaher, what do you think about modifying all of our code to use just a normal image message type, then setting up a simple node that gets the wfov_image and republishes the normal image?

PR #49 (2017-01-15): Fisheye Camera additions as well as bottom camera
Made fisheye closer to real cameras and added bottom camera. Fixes #16 and #40.

---
This change is [[image]](https://reviewable.io/reviews/palouserobosub/robosub_simulator/49)
closed 2017-01-15 by bkallaher