PalouseRobosub/robosub_simulator-2018 #20: Add hydrophone period parameter

ryan-summers, 2016-12-13:
Current, the hydrophone time deltas are published every cycle that gazebo sends out position messages, however this is not close to the real situation. In reality, hydrophone pings only occur once every two seconds.
On the other side of the coin, high rates are desirable for debugging systems to ensure that things are working properly. For this reason, add in a parameter for the rate at which hydrophones will ping. This could then be loaded by the simulator bridge so that it only sends out a time delta message at that rate.

bkallaher, 2016-12-13:
Use the ThrottledPublisher class.

BigDCross, 2016-12-14:
Is there any documentation for the ThrottledPublisher class @bkallaher ? Because I can't find a thing...

skallaher, 2016-12-14:
The ThrottledPublisher is a class I made at James's request. It should be under src/utility.

irwineffect, 2016-12-15:
Hmm...to properly use the ThrottledPublisher class, it looks like we need to tweak our settings so that our package is properly exported, then have the simulator require the package. Some relevant links:

http://answers.ros.org/question/201977/include-header-file-from-another-package-indigo/
http://docs.ros.org/indigo/api/catkin/html/howto/format2/building_libraries.html

BigDCross, 2017-02-04:
Fixed by #52

closed 2017-02-04 by BigDCross