PalouseRobosub/robosub_simulator-2018 #47: Add Marker Drop

BigDCross, 2017-01-15:


brennanpeter, 2017-02-23:
James suggested I try to help you out on this @BigDCross Where do I get started?

BigDCross, 2017-02-24:
@brennanpeter I would write a model plugin for gazebo that we can attach to the sub model. Within plugins in gazebo you can programatically spawn models so it would could just subscribe to a marker drop topic (with a custom marker drop message maybe? Or since we'll have to spawn torpedoes as well maybe a message for both?), then spawn the model at the appropriate location. I'm not sure you can set velocity directly while spawning but you can always set the velocity after spawning the marker/torpedo.

Here's a few resources to get you started:
**Inserting Models:** http://gazebosim.org/tutorials?tut=plugins_world (You'll need to get the world ptr for this to work. You can do that with this method: https://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/classgazebo_1_1physics_1_1Base.html#a9e27deb7b7cc4bf50f8625edcb9c38ad)

**Model Plugins:** https://github.com/PalouseRobosub/robosub_simulator/blob/master/examples/model_plugin_example.cpp

**ROS Subscribers within gazebo plugins:** http://gazebosim.org/tutorials?tut=guided_i6

**Setting model pose:** https://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/classgazebo_1_1physics_1_1Entity.html#ac8eefcc7635536d1b2d95275f04a9b1f

closed 2017-04-15 by bkallaher