PalouseRobosub/robosub_simulator-2018 #67: Add realistic noise

edwardwsu, 2017-02-27:
Here's the values that was retrieved from the bag and in lab testing. Can we add it to the simulator to tune to sub?

Data from sitting at surface: 

depth avg : -0.0397458875721
variance : 1.14782991135e-05
Z avg: 0.05415
Z variance: 0.0021462775
Roll avg : -0.0320515607958 
Roll variance : 0.000242492278412
Pitch avg : 0.0426270517613 
Pitch variance : 1.54346580708e-06
Yaw avg : 0.248215400555 
Yaw variance : 0.00584271433361

Data from sitting on lab cart:

depth avg : 0.306843254783
variance : 8.68346523759e-06
Z avg: 0.100454545455
Z variance: 0.00173657024793
Roll avg : 1.62338589188 
Roll variance : 9.11627383596e-29
Pitch avg : 0.227304441355 
Pitch variance : 2.49600520793e-31
Yaw avg : -2.83088092224 
Yaw variance : 1.23259516441e-28

skallaher, 2017-02-27:
Which set of data do you propose we use? I assume this is to model the sensor noise and I don't know how you would use both at once. Would this still remain valid when the sub is moving?

irwineffect, 2017-02-28:
We're not really sure which would be the best to use, let's just start with the data of it sitting on the lab cart.

btmoore, 2017-03-01:
There isn't a point to adding these values to the simulator.   Look at the values you're getting.  e-6, e-31, ... There's no noise here.  Like, literally bottoming out on floating point granularity.  It will make sense to add noise to the virtual system, but not with these numbers.

Incidentally, when you do, it's going to be a long process. You're going to have to make a separate module that makes up simulator-derived measurements, adds noise to those measurements, runs them through the localization module, and then reports _that_ to the control system.

As a more general point, the simulator needs _realistic_ values, not _real_ values.  If you make everything perfect to your estimates of noise, all that does is validate your system if your estimates are good.  More generally what you want, is to just assume an upper bound on your noise (and then go a bit above that) and then see if the system still functions.  That's how you validate the system. By giving it worse-than-worst case scenarios and seeing if it still functions.

irwineffect, 2017-03-02:
Ahhk, so we can take those variances and multiply them by 100. Although, I don't see why adding the noise would be a long process, all we have to do is run the simulator measurements through random Gaussian generators right?

btmoore, 2017-03-03:
First, we're not talking x 100.  we're talking like, literally times 10^28 for some of those.  Angle error is probably going to be about a 2 or 3 degree variance when everything is said and done.

But that aside, what you can do is you could add noise on top of the exact submarine state that is reported to the control system.  But the noise levels you choose for those are not going to be based on the sensor noise.  Because the sensor readings, with their noise, get fused together and are constrained by their past readings.

So the yaw reading could be jumping like, back and forth at +/- 5 degrees, and if the submarine took those readings as gospel it'd be wagging its butt all the way - it would never hold a steady course while moving forward.  With the filters in place that might get down to more like +/- 1 or 2 degrees, and more importantly any dramatic jumps would be smoothed out so the behavior wouldn't be as jerky.

So, you could add noise to the simulator, but the noise you add will be neither the sensor values you've posted above, nor the noise of the sensors themselves.  It will have to be lower to get realistic behavior, otherwise it'd be too jerky.

ryan-summers, 2017-03-06:
If anyone is interested, this now exists on rsummers/noise. I need to update the branch to use a different variance for the Z linear acceleration because the BNO specifies the variation of Z differently.