PalouseRobosub/robosub_simulator-2018 #81: Modify Image Publishing to output WFOVImage messages on correct topics

skallaher, 2017-03-18:
Change the simulator to publish a WFOVImage rather than an Image to imitate the pointgrey camera drivers. 
Change topics published to for these images to be **/camera/{left|right|bottom}/image** rather than **/camera/{left|right|bottom}** to replicate pointgrey camera drivers.

irwineffect, 2017-03-20:
Note that if we complete PalouseRobosub/robosub#221, then there is no need to have the simulator publish WFOV. Publishing a normal image instead of WFOV image is the most desired behavior.

skallaher, 2017-04-14:
Looking at the sdf, it would be much easier if the cameras published a standard image as making the simulator publish a WFOV would involve messing with the gazebo_ros_camera library.

skallaher, 2017-05-12:
This is not needed after #93 and PalouseRobosub/robosub#246 are merged.

skallaher, 2017-08-26:
Fixed via PalouseRobosub/robosub#246.

closed 2017-08-26 by skallaher