PalouseRobosub/robosub_simulator-2018 #99: Gate task unit test

Sindorman, 2017-10-09:
Implement a way for either simulator to detect when sub passed the gate or make a unit test implicitly detect it based on it's location/position.

skallaher, 2017-10-09:
This issue belongs in the Robosub repo.

closed 2017-10-09 by skallaher

Sindorman, 2017-10-10:
@skallaher I haven't researched it yet, but I expect it to involve changes in simulator. For example a script that reads/detects position of sub in simulator, and checks it relative to the box of exit side of gate.

skallaher, 2017-10-10:
The test itself can listen to the output from Gazebo and get the position of any and all models in the environment. This includes the gate and submarine. There should be no need to make changes to the simulator.

Sindorman, 2017-10-11:
But it will be probably needs to be placed in robosub_simulator repository, because it will not run on the sub itself?

skallaher, 2017-10-11:
There isn't a clear cut answer to this. Having the tests in the robosub repo also makes sure that they are available to run when the build server is run. In addition, this keeps the tests near the source they are testing. From what I can tell, unless @irwineffect knows otherwise, this is merely a design decision.