PalouseRobosub/robosub_simulator-2018 #119: Dice are not being displayed properly.

Sindorman, 2018-02-01:
Me and Lance have noticed that Dice are not being loaded fully. Here is an image of what they look like.
simulator branch is `master`.
[image]

ryan-summers, 2018-02-01:
@irwineffect I think we need to add Git LFS as a dependency of the robosub packages to ensure that people have it installed on their system. I think this is because the JPEG textures aren't getting pulled down because LFS isn't installed on the target machine.

ryan-summers, 2018-02-01:
I _believe_ this should be corrected by typing `git lfs install` in the simulator repository and then pulling master.

neprotalis, 2018-02-01:
This is correct. I talked with Sean about it and he is going to show me how to add it to the repo. It will be corrected soon. If you need it now though https://github.com/git-lfs/git-lfs/wiki/Installation

ryan-summers, 2018-02-01:
After LFS is installed, type `git lfs pull` to pull down the JPEG textures.

closed 2018-02-01 by ryan-summers

hmmwhatsthisdo, 2018-02-01:
@ryan-summers `git lfs install` only works once the `git-lfs` package is added, unfortunately.

ryan-summers, 2018-02-01:
We're adding it to robosub package dependencies.